gauteh/sfy #31: supercaps smart module

jerabaul29, 2022-02-17:
tracked here: https://forum.sparkfun.com/viewtopic.php?f=103&t=57215 .

No answer so far, if you want you can +1 on it :) .

gauteh, 2022-02-20:
How big are the capacitors on the AGT? Is it two in series to get to 5V (these ones: https://www.sparkfun.com/products/746?)?

@jthomson-apluw, alex: I noticed you have some large capacitors on the microSWIFT: are those for the Iridium modem? Do you charge them in a special way (or limit current with resistor), or are they connected directly over the batteries?

jerabaul29, 2022-02-20:
I think having 2 supercaps in series without anything more may calls for problems, as they are not perfectly identical, so there will be some unbalance developing with time as they are used and current flows...

I think that the AGT actually uses only 2 small supercaps by default, 2x1F2.7V (see https://cdn.sparkfun.com/assets/1/5/e/f/4/SparkFun_Artemis_Global_Tracker.pdf). If I remember well this is enough to limit peak battery draw to 350mA or something like this. To limit even further, it is possible to solder 2 10F2.5V supercaps (these are the ones you mention, but not in place by default).

I think a possible solution with minimum extra circuitry in your context could be:

- use only 3 batteries in series, i.e. 4.5V nominal, 5.1v max, possibly several such packs in parallel of course :)
- use one or more supercaps rated for 5.4V in parallel of this: https://www.digikey.no/no/products/detail/SCMT32D755SRBB0/478-SCMT32D755SRBB0-ND/8028702?utm_medium=email&utm_source=oce&utm_campaign=3319_OCE21RT&utm_content=productdetail_NO&utm_cid=470231&so=71613812&mkt_tok=MDI4LVNYSy01MDcAAAF_XmkxvBBe6woDYDGDsty0Co2tgn1iTsnaJ6C4nn2uoDBp-qlE8qILHARJ9kw8_gfIr4E677i8uF7NEgDPtEomt07m8NfM7iV8YcOHOa8 or https://www.digikey.no/no/products/detail/SCAP%2cPBLL-15.0%2f5.4/2085-SCAP%2cPBLL-15.0%2f5.4-ND/9929727?utm_medium=email&utm_source=oce&utm_campaign=3319_OCE21RT&utm_content=productdetail_NO&utm_cid=470231&so=71613812&mkt_tok=MDI4LVNYSy01MDcAAAF_XmkxvOPGaTiLq8EJE4KahpRQMIIJkKLbbU9Lm-SuSyGwklI4s6zWwhKTx7LxdKz2WuFdDXQfMXL46FLvpvk8XeGUBPh7Za4tGr5dNU4 for example

This way you avoid the need to have 1 supercap in parallel of each cell, or a dedicated supercap balancing chip, and you can have just a few supercaps added on your whole power supply pack. I see your point that if batteries get low the voltage drops, but I think the supercaps can keep it high enough that the pololu does not switch off?

jerabaul29, 2022-02-20:
I actually had bought privately a couple of these to test them; you could get one of each next time we meet if you want :) .

gauteh, 2022-02-21:
Thanks. It should be [OK](https://www.sparkfun.com/products/746) to put them in series to increase voltage capacity, it will half the capacitance though! I will test:

* Use 1x6 D-cells + [S13V30F5](https://www.pololu.com/product/4082) to see if batteries perform better in this case.
* Put capacitors in series over entire battery pack (2x3 D-cells), I don't think putting them over each cell gives a significant advantage over the extra wiring. This will introduce 1) initial short (limiting resistor of even 1Ohm would produce high voltage drop when pulling 2A later -- but haven't simulated or though very hard about this), 2) current-leakage (maybe not a significant problem since I am sending so often).

jerabaul29, 2022-02-21:
Yes, will be very exciting to hear how it goes :) . You could set up a small 'supercap pre changing bench' and / or connect tte batteries initially through a 10ohm resistor and then remove it :)

jthomson-apluw, 2022-02-27:
We use the large capacitors to handle any [brief] interruptions in power supply from the batteries when we drop these from the air…

gauteh, 2022-03-04:
Thanks.

Asked Pololu for some advice: https://forum.pololu.com/t/powering-a-gsm-modem-with-alkaline-batteries/22758

Cellular modem producer recommends lithium, but want to stay on alkaline for as long as possible: https://discuss.blues.io/t/powering-notecard-using-d-cells-2x3-or-1x6/703

Now testing 6x1.5V (D-cells) through [S13V30F5](https://www.pololu.com/product/4082). Regulator is audibly working when modem is transmitting. Will see how low current it can keep operating to.

It is possible that 4.5V with 2x3 D-cells work better, since the modem can be powered directly from the batteries. The S13V30F5 seems to struggle delivering enough current when input voltage is too low.

gauteh, 2022-03-11:
Turns out the biggest problem was that the wires and breadboards were not sufficently rated for conducting the required current. Hooking things up as directly as possible seems to work better. There was also a bug in missing delays in transmitting to the modem, which when fixed reduced communication errors.

https://discuss.blues.io/t/notecard-i2c-error-not-recoverable-cant-connect-stuck-in-extended-network-failure/670/29?u=gauteh

Very good help from the blues.io team!

closed 2022-03-11 by gauteh