gauteh/sfy #35: IMU: use continuous FIFO mode, not FIFO mode

gauteh, 2022-02-25:


jerabaul29, 2022-02-25:
what is the difference? :)

gauteh, 2022-02-25:
Minor difference, ST suggested I do this. I don't think there is a
difference now, but it should be in continuous mode for what I am trying to
do. Otherwise the FIFO needs to be reset every time I drain it. But then I
would loose a few samples.

gauteh, 2022-03-01:
Its fine to use it as it is: https://community.st.com/s/case/5003W00000FWwfEQAT

closed 2022-03-01 by gauteh

jerabaul29, 2022-03-01:
Linken funker ikke, har du en annen link / navn på selve tråd så at jeg kan søke den? :) .

gauteh, 2022-03-01:
Må kanskje være logga inn med min konto:

[Case 00152476.pdf](https://github.com/gauteh/sfy/files/8161418/Case.00152476.pdf)

jerabaul29, 2022-03-01:
Jeg har registrert meg, og er logga på. Rart. Men takk :) .