gauteh/sfy #48: precompiled standard binaries?

jerabaul29, 2022-04-22:
@gauteh do you think you could provide some standard pre compiled binaries, that can be uploaded with the SF GUI tool, similar to what I do on the OMB? :) .

gauteh, 2022-04-22:
Yes. The problem is that you need to specify the account the data is going to in the binary. So I haven't figure out how to do that yet.

jerabaul29, 2022-04-22:
Ah, ok. Would it be possible to have something like (of course this is only my 2 cents, may be naive / flawed, just thinking loud):

- by default, a space in non volatile but writable memory (Flash is ok for that, and I think it should be what is used, as there is no EEPROM on the artemis boards / ambiq chips) is initialized to all 0s and a "flag account is set" is set to false

Then at each boot:

- if the flag is set to true and the account updated, then work as normal with the account that is registered in Flash
- if not, open a small USB-over-serial based interface that ask for entering the account (with double confirmation, yes/no confirmation etc), and upon account successfully entered, save the account information in Flash, set the flag to true, and start "normal" operation.

It could also be possible, at each boot, to allow for a "grace period" of 10s or so when the instruments just waits a bit and the USB-over-serial interface is activated, so that even already configured sensors can be re configured.

This way the user could "simply" upload over USB with the SF GUI tool your firmware, and then configure it over USB at the first following boot. Or am I missing something? :)

gauteh, 2022-04-22:
That should work :)

gauteh, 2022-05-16:
#54 mitigates this issue somewhat, will make it easier to build the binaries independent of environment.

gauteh, 2022-10-11:
One solution is to set the buoy-id and product through the notehub, but that requires a step similar to rockblock where you assign or register the device. Can maybe be made optional.

gauteh, 2023-09-29:
https://discuss.blues.io/t/register-notecard-with-product-and-name-directly-in-the-notehub-or-outside-the-card/1690

commit 649c964a (2024-09-10): build firmware (fix #48)
commit 291eaa43 (2024-09-10): build firmware (fix #48)
commit b938f7a1 (2024-09-10): build firmware (fix #48)
closed 2024-09-18 by gauteh