gauteh/sfy #53: SD card: store data

gauteh, 2022-05-16:
In progress.

* https://users.rust-lang.org/t/wave-buoy-storing-data-to-sd-card-robustly-and-transmitting-over-unstable-connection/75512

gauteh, 2022-05-16:
Yes. Although FRAM is a better solution for completely embedded storage, I think SD-card is more practical in out case since:

* the buoys are generally recovered, but electronics might be damaged by water
* FRAM requires an online buoy, or a recovery-station
* I suspect that SD-cards are more robust against water than FRAM
* SD-cards can be easily read from a computer as long as a standard file-system is used

jerabaul29, 2022-05-16:
I am not sure about SD card being more robust than FRAM. I would think the contrary, as it would be possible to "hot gun glue" the FRAM chip to protect it 100% from salt water, and "hot glue gun" some soldering pins on the FRAM breakout in the same way. If there is salt water, all what is needed is melt the glue on the soldering pins, and get back access to the traces through pads that were not corroded. But of course only my 2 cents, may be wrong.

zerblatt007, 2022-05-18:
Maybe try covering the SD connector in Super Lube PTFE oil for protection
against the elements?
We use it in Netatmos outdoor cameras to secure a good connection. No
problem using the connector with oil on it in readers..

https://helpcenter.netatmo.com/en-us/smart-outdoor-camera-with-and-without-siren/product-interactions/how-is-my-smart-outdoor-camera-protected-against-rain-and-humidity
https://www.3dkongen.no//produkt/12g-super-lube-multi-purpose-synthetic-grease-with-syncolon-ptfe/

gauteh, 2022-05-18:
Thanks, that will be very useful.

closed 2022-05-29 by gauteh