gauteh/sfy #62: dashboard: adding a "display trajectory" set of features for buoys

jerabaul29, 2022-06-07:
Not at all urgent of course, more of the "brainstorming" kind, possibly good tasks if some more people join the project some day (extending the dashboard sounds like a really nice project for some BSc students or something like this; we have some students coming to IT now and then, I could keep these features in the back of my head and suggest them as a topic for students to work on as a task if you want :) ).

Would be great to have:

- a tick box for each entry in the "Devices overview" to show / hide the full trajectory
- "calendar choice boxes" to be able to also specify a dates interval over which showing the trajectory
- if "latest" is used as the "right end" of the time interval to plot, and a buoy has not transmitted for 12 hours, put a cross at the end of the trajectory
- clicking on a buoy symbol make the trajectory become bold / highlighted
- when there are "holes" of more than 3 hours, the "straight line" becomes dashed, to indicate the dropout that takes place
- the server remembers the last set of settings used for each buoy, so that the settings persist from connection to connection
- the server, even for the same data, allows creating several "profiles" or "user profiles" that allow to display data with a given set of options, and to remember these options, so that easy to plot again and again in the same way, and switch between ways of plotting

gauteh, 2022-06-07:
Sounds good! A re-write to react / ts would also be fine, might be more easier to find people for. Current framework is pretty similar to react. And maybe move to https://parceljs.org/ for building.

Simple configuration can also be stored in cookies.

jerabaul29, 2022-10-14:
- possibility to display wave data: show SWH and Tp and spectra multiplots by selecting a set of buoys and clicking on a "render wave data" button

gauteh, 2022-10-14:
👍 we're demoing the buoys at the university karrieredag next week, and something like this would be ideal. But I don't think we'll be able to get it ready by then.