gauteh/sfy #79: NAK / SwError / ARB with IMU: add pullups to SDA / SCL

gauteh, 2022-08-14:
SDA and SCL should have pullups to VDD_IO, but those are currently not installed. I don't think there are any (external) pull-ups on the artemis either. Check if adding 10k pull-ups (especially when running at 1MHz) helps.

@jerabaul29 : Might want to consider this if you use the STEVALs. The pull-ups might be present on the adafruit / sparkfun boards.

jerabaul29, 2022-08-14:
Thanks for the tips. Do you think this causes some errors?

gauteh, 2022-08-14:
I'm seeing NAK or ARB errors, especially when voltage is lower. It recovers and boots up fine again immediately. But both these could be caused by missing pull-ups. Especially at high-speed communications. The ARB error sometimes stems from timing issues, but this is inside an interrupt handler, so I don't think that is the issue.

jerabaul29, 2022-08-14:
@TryggviLab @freeka100 @T-Kodaira @tnn77 possible improvement: adding the pull-ups ?

gauteh, 2022-08-14:
The issue happens with around 20 minutes intervals:

```
2022-08-14 17:08:23.670000+00:00  2022-08-14 17:30:24.783077+00:00       1260.31    0:21:00.307692          0:00:02.699230      0:00:21.157692          64       10070     10133
2022-08-14 17:30:50.380000+00:00  2022-08-14 17:31:52.643077+00:00         59.0769  0:00:59.076923          0:00:02.297692      0:00:25.596923           3       10134     10136
2022-08-14 17:32:09.540000+00:00  2022-08-14 17:54:14.643077+00:00       1260.31    0:21:00.307692          0:00:04.397692      0:00:16.896923          64       10137     10200
2022-08-14 17:54:36.540000+00:00  2022-08-14 17:56:40.643077+00:00        118.154   0:01:58.153846          0:00:03.038461      0:00:21.896923           6       10201     10206
2022-08-14 17:56:53.070000+00:00  2022-08-14 18:17:09.702308+00:00       1161.85    0:19:21.846154          0:00:02.849231      0:00:12.426923          59       10207     10265
2022-08-14 18:17:32.400000+00:00  2022-08-14 18:39:38.683077+00:00       1260.31    0:21:00.307692          0:00:06.236923      0:00:22.697692          64       10266     10329
2022-08-14 18:40:09.110000+00:00  2022-08-14 19:02:14.073846+00:00       1260.31    0:21:00.307692          0:00:04.616154      0:00:30.426923          64       10330     10393
2022-08-14 19:02:40.290000+00:00  2022-08-14 19:24:38.803077+00:00       1260.31    0:21:00.307692          0:00:02.878461      0:00:26.216154          64       10394     10457
2022-08-14 19:25:05.520000+00:00  2022-08-14 19:47:46.002308+00:00       1299.69    0:21:39.692308          0:00:02.868461      0:00:26.716923          66       10458     10523
2022-08-14 19:48:30.900000+00:00  2022-08-14 19:51:37.283077+00:00        177.231   0:02:57.230769          0:00:02.878461      0:00:44.897692           9       10524     10532
```

that's the time between transmissions. The transmissions can cause voltage drops. Maybe not fixable, and power is just low, but worth adding the pull-ups.

tnn77, 2022-08-15:
@jerabaul29 is this only when using STEVALs?

jerabaul29, 2022-08-15:
@tnn77 yes, I think only for STEVALs.

Also, as @gauteh shows in his last answer, it looks like this happens possibly due to voltage drops on his setup that is using a GSM modem that uses a lot of current - so not sure that adding the pullups would help anyways.

All in all, it is likely that the setup we have is fine, but cannot hurt to add the pullups if there is possibility to add them easily on a PCB :) .

gauteh, 2022-08-15:
They are required by the ISM330DHCX datasheet.

closed 2022-09-23 by gauteh