gauteh/sfy #86: PCB-fixes for v0.3

gauteh, 2022-09-23:
- [ ] 10kOhm pull-up on SD MISO
- [ ] 10kOhm pull-up on SD MOSI
- [ ] 10kOhm pull-up on SD CS (chip-select)
- [ ] approx 50 uF (tantalum?) capacitor on SD power-supply
- [ ] Move card-holder?
- [ ] Larger screw-holes
- [ ] Larger +BATT / -BATT holes
- [ ] Move switch-holes

gauteh, 2022-09-23:
[image]

with pull-ups.

jerabaul29, 2022-09-23:
Funker også SD kort med ekstra komponenter rundt? :)

jerabaul29, 2022-09-23:
Det er bare brainstorming, og godt mulig du kan det mye bedre enn meg :) (hvis det, sorry for at jeg bare skriver åpenbare ting her ^^ ). Men angående SD-kort: det er ganske høy frekvens på de signalene som går gjennom SPI. Tror da at det kan vært viktig å gjøre sikker at:

- det er så korte PCB traces som mulig
- det er ikke noe "current loop", ie GND følger tett på +3.3V og signalene, og GND kommer tilbake gjennom spesifiske PCB traces som følger +3.3V og signalene, ikke gjennom en base plate som kan gi en ganske annerledes vei til tilbakestrøm
- CLK og DAT er nære hverandre

gauteh, 2022-09-23:
Gode forslag, lurt å sjekke hvis vi rekker før vi bestiller neste. Funker
tilsynelatende bra med motstander.

jerabaul29, 2022-09-23:
Aaah, så kult at det funker, gratulerer :) . Da funket alt på PCBen med et par ekstra komponenter stemmer det? :)

gauteh, 2022-09-23:
Ja, litt ting som må gjerast meir praktisk som du ser. Men ellers ser det
bra ut!

zerblatt007, 2022-09-24:
MET logo er vel smart? God plass der SD kortet stod?

gauteh, 2022-09-24:
Blir det under notecarrieren? Sikkert ikkje so dumt med ein URL eller ein eller anna kontakt-info.

gauteh, 2022-09-25:
La til ein MET logo og ein link, eg antar det hullet framfor SD-kortet er for å låse det? Ellers hadde det blitt litt meir plass.

zerblatt007, 2022-09-25:
Har flyttet skruen litt, men logo og link ser jo gei ut, men er jeg.. :)

zerblatt007, 2022-09-25:
Da er jeg vel så fornøyd jeg kan bli akkurat nå, tror jeg.
Labling var lurt for da ser man hva nettene heter i PCB mode, men jeg har ikke gjort det på alt.
NC greiene får jeg ikke til å fungere, så det gir jeg opp nå, men tror alt skal være koblet riktig.
Jeg la på en rev2022-09 tekst på baksiden som jeg tenkte kunne være lurt også. Hadde håpet på å kunne bruke en variabel i teksten, men fant ikke noe sånn fort, så jeg skrev det inn manuelt.

gauteh, 2022-09-25:
Ja, dette ser bra ut. Da er vi vel i grunnen klar til å bestille ei ny runde? Kan la det ligge å surre litt til i kveld eller i morgon før eg gjer bestillinga i tilfelle vi tenker på noko.

gauteh, 2022-09-25:
Da har eg bestilt eit nytt sett. Sidan vi kun har komponenter på ei side vart det noken hakk billigare. Det var minimal pris for å bestille fleire stk (0.20 USD per prett utover 5), so eg bestilte 15. Vart 79USD inkl assembly og frakt.

jerabaul29, 2022-09-25:
Det blir spennende :) .

Ser litt på EasyEDA vs KiCAD, er litt i tvil: det er litt synd med noe close source nettbasert løsning, men jeg må si at det høres kjempefint ut å ha en CAD program som funker "out of the box" med "library parts" og kan bli brukt til å bestille med bare en click :) .

gauteh, 2022-09-25:
Ja, det har sine bakdeler. Det er utrulig strømlinjeforma.. tar ti minutt til det er bestilt i frå eit ferdig PCB brett. Du kan eksportere til Altium og det finst konverter til kicad frå easyeda arkiv.

closed 2022-09-26 by gauteh