gauteh/sfy #118: move sfy-data to kubernetes (or something)

gauteh, 2023-03-14:
With #114 it is possible to set up sfy-data with postgres and kubernets. we're hitting some limits w.r.t scaling and backup of disk-space. the server is stateless so it is possible to have multiple servers. @cskarby started the process of setting this up, and I have also not had any time to follow up yet. @jerabaul29 do you have anything to do with kubernetes?

jerabaul29, 2023-03-14:
I am involved a bit in a project that uses k8s on Azure. I am by no way an expert, but would be interested to learn more. Not sure with what hours / when I could find the time though (but that could be a side project for the summer or something like this).

jerabaul29, 2023-03-17:
(I will try to discuss this next time I have a planning meeting, I would really be interested to look a bit deeper into this, would be really great for me to learn more about these technologies :) ) .

gauteh, 2023-11-07:
Sounds good!

closed 2024-09-10 by gauteh