gauteh/sfy #124: Reduce power use

gauteh, 2023-04-25:
We have not spent any significant time trying to reduce power use, and it is currently on 50 mA @ 4.5V on avarage which is higher than expected. Especially when trying to set up a long term system for #123 this would be great to put some effort into.

A few initial points to start with:

* Configuring the notecard: https://discuss.blues.io/t/notecard-not-entering-low-power-mode/1035/7
* Do not communicate with notecard unless necessary (not every loop)
* Cut LEDs
* Measure Artemis and Notecard individually to figure out where the current is drained.
* Do not send packages to notecard immediately, but cache to other memory or SD card.
* Maybe SD-card is the big drain, maybe not necessary.

gauteh, 2023-04-25:
Maybe @akkermei has some suggestions.

jerabaul29, 2023-04-25:
LEDs are surprisingly hungry in my experience on the OMB. Another point is how you feed power to the chips - using the pololu step up step down and feeding directly on the 3.3V power rail, bypassing regulators etc, may help quite a bit, if this is feasible.

An extra option that is possible is maybe to reduce the duty cycle? 20mins every hour for example? That would be an easy 2/3 power reduction? :)

See the typical OMB power consumption in https://www.researchgate.net/publication/357712696_OpenMetBuoy-v2021_an_easy-to-build_affordable_customizable_open_source_instrument_for_oceanographic_measurements_of_drift_and_waves_in_sea_ice_and_the_open_ocean table 1.

gauteh, 2023-04-25:
It is only one regulator at the moment, the V+/GND on the Notecarrier. The rest is powered from VIO on the notecarrier to 3V3. So it should be pretty similar to having a Pololu power 3V3.

I suspect the SD-card might pull things up a fair bit, it is only used every 20 seconds, but maybe it should be powered down between writes.

gauteh, 2023-06-24:
Now using between 40 to 80 mW.

[image]

closed 2023-06-24 by gauteh