gauteh/sfy #139: Fuse GPS and IMU to get accurate position

gauteh, 2023-06-27:
E.g.: https://github.com/ydsf16/imu_gps_localization

jerabaul29, 2023-06-27:
This is definitely the way to go long term - I have been wanting to look into this and "do it properly" for a long time, but never found the time / would need a large project to be able to do it. Likely developing a custom Kalman filter with acc + gyr + GPS_pos + GPS_doppler would be the way to go. There are very good resources available online for learning Kalman filtering - one of the best I know is https://github.com/rlabbe/Kalman-and-Bayesian-Filters-in-Python and refs therein.

gauteh, 2023-06-27:
I wonder how long the gyro is stable, or whether we need a magnetometer as well. Running the GPS is a bit power-expensive. Is it one minute, ten minutes?

jerabaul29, 2023-06-28:
The gyro is not "stable" in the sense of the "old" mechanical rotating gyros; it has a high dynamic sensitivity but may have quite some drift (especially uncalibrated). The ISM330DHCX is not bad though, see: https://learn.adafruit.com/adafruit-sensorlab-gyroscope-calibration/comparing-gyroscopes .

What happens is that the Kalman filter uses the fact that "in average, acceleration is downwards" to get absolute information relative to the vertical to avoid drift of the estimated vertical over time. If the magnetometer can be got to work well (i.e. need calibration, far from batteries and currents and magnets etc), the same could in theory be done for orientation relative to N (though this is problematic in the polar regions, as magnetic field is nearly vertical there so compasses and mags do not work well). The best would be to use GPS doppler and compare it with accelerations in horizontal plane, though I agree this is power hungry. May be possible to take down the GPS sample rate though maybe to save a bit of current (I do not think horizontal direction information is needed at more than 1Hz typically, not need to have high frequency 10 or 20Hz GPS data).

We can chat more about this some day if you want :) .

jerabaul29, 2023-06-28:
Btw, the newest GPS modules should have quite a bit lower power consumption. Traditionally, GPS modules have had around 30mA consumption, but the newest series should go down to around 10mA in continuous mode (less in non continuous mode / using fewer satellite constellations). I think a power consumption of 30-35mA for 20 mins is a bit "expensive" for a low power buoy, but 10-15mA may be quite ok.

The GPS module I think about:

- datasheet: https://cdn.sparkfun.com/assets/7/5/9/a/a/MAX-M10S_DataSheet_UBX-20035208.pdf
- brief: https://content.u-blox.com/sites/default/files/MAX-M10_ProductSummary_UBX-20017987.pdf
- a few more words about it: https://blog.adafruit.com/2022/10/06/eye-on-npi-u-blox-max-m10-series-gnss-modules-eyeonnpi-digikey-digikey-ublox-adafruit/
- SF product: https://www.sparkfun.com/products/18037

It could be a good project for some student to look into this module in details and see how low power consumption is attainable in practice in different modes :) .

jerabaul29, 2023-06-28:
@prbush this is a point I would love to hear your experience about (the low power GPS modules aspect). Which GPS module are you using / how low continuous operation power use do you reach? :) ).

gauteh, 2023-06-28:
Yes, that should be fine for the OMB. I think it will be too hungry for the SFY for long-term deployments. No problem for short ones, but otherwise we can do full-GPS for a limited time and then use lower-frequency GPS in-between.

prbush, 2023-06-28:
We are using the same chip you mentioned above, the Max-M10S. Great chipset. We run 5 Hz continuous mode. I don’t think we can make use of cyclical tracking mode for our application. UBlox has almost too many configuration options, including a lot of binary encoded, proprietary message types; we are using one to get velocities. I like their messages a lot more than working with NMEA sentences.

I have not yet profiled the power consumption of this chip. I’m going to be getting with our electrical engineer at some point in the near term to profile the power consumption of our system. I’ll take measurements of the GNSS chip and let you know what real world values look like.

jerabaul29, 2023-06-28:
That will be very interesting :) . I heard antenna can modify a lot power consumption, do you use an active or passive antenna?

prbush, 2023-06-28:
We are using passive. Our current antenna configuration leaves a lot to be desired and will be changed. It’s not easy to get great signal quality on the waterline. Using an active antenna is on the table, but only as a last resort. We use a circularly polarized antenna tuned for both GNSS signals and Iridium, with an RF switch to port between the modem and GNSS chip.