gauteh/sfy #170: RTK+PPP

gauteh, 2024-09-10:
* [ ] Configure and log RTK+PPP (@edwinrainville) - test with openlogger
* [ ] Get access to ublox PointPerfect (@edwinrainville any news on this?)
* [ ] Read and transmit ublox telegrams using sfy (@gauteh)
* [ ] Add stronger power regulator (Pololu?) - already at max (#169).

* https://www.sparkfun.com/products/22560

gauteh, 2024-09-10:
Currently we use the regulator on the notecarrier. Reluctant to use the artemis one since it appears to be less efficient (and it cannot take low voltage).

edwinrainville, 2024-09-10:
I have the PointPerfect account set up from ublox and got us a 2 month free trial of the L-Band broadcasted PPP correction data.

Here is also a link that the support staff sent me on the correct configuration of the combo ZED F9P and NEO D9S board. https://developer.thingstream.io/guides/location-services/pointperfect-getting-started/pointperfect-l-band-configuration

I was also following along with an example setup I found for the combo board and was looking into how to modify this to allow us to log the correction data. This example code configures the combo board through a qwiik connection with another board such as the Artemis nano I have been playing with. It is a little unclear to me if the configuration firmware is saved on the combo board or the main logger using this version. While I think having a firmware code that we can version control and reproduce for other buoys - a lot of the examples suggest using the U-blox platform to set it up.

gauteh, 2024-09-10:
Sounds great! I think for playing around with the configurations and figuring out what is good, the ublox platform is nice, but at some point we need to send the equivalent parameters over the firmware. We could probably even skip that for now (if we don't have time) as long as the configuration is saved persistently to flash or similar.

That means we run of of the subscription before the experiments, have to remember renewing that before we go.

gauteh, 2024-10-02:
* Connecting A0 to ext-gps-mod TX
* Connecting A16 to ext-gps-mod RX
* Connecting A2 to PPS

gauteh, 2024-10-02:
@edwinrainville we need to figure out if the F9P produces PPS every sample (20 Hz) or every 1Hz, otherwise I need to read a different number of telegrams on each pulse. I am guessing/hoping every 20 Hz. That is probably also the reason they named in time pulse (TS) over PPS.

edwinrainville, 2024-10-02:
I'll see what I can find in the documentation on how the TP setting is configured.

gauteh, 2024-10-03:
You can just try to set it here: https://github.com/gauteh/sfy/blob/main/sfy-buoy/sfy-ext-gps/ext-gps-mod/gps.cpp#L31 (to 20 Hz) and see what happens.

gauteh, 2024-10-03:
[sfy1 ext-gps.pdf](https://github.com/user-attachments/files/17244320/sfy1.ext-gps.pdf)

gauteh, 2024-10-19:
I'm trying to calculate ranges for serializing the lon and lats into u16s (https://github.com/gauteh/sfy/blob/main/sfy-buoy/src/gps/wire.rs#L19):

```
// The distance movable within one packet length of maximum 20 seconds. For 100 km/h this is
// (100 * 1000 / (60 * 60) * 20 ~= 550 m
const MAX_KM_PER_DEGREE: f32 = 111.3;
const DEG_PER_M: f32 = 1.0 / (111.3 * 1e3);
pub const LON_RANGE: f32 = 2.0 * DEG_PER_M * 550.0 * 1e8;
// pub const LON_RANGE: f32 = 2.0 * 1.0e8 * 550.0 / (MAX_KM_PER_DEGREE * 1.0e3); // 550 m in both directions
                                                                        // [deg * 1e-7]
// Maximum distance within 20 seconds: 60 m ?
pub const MSL_RANGE: f32 = 2.0 * 60.0 * 1.0e3; // 60 m in both directions [mm]
```

It seems that 1e8 of a degree is 1.113 mm? Assuming a worst case scenario of 111.3 km per degree. Is my thinking right? With a maximum of 550 m movement in 20 seconds (100 km/h over 20 seconds) this gives a worst-case resolution of 16.8 mm.

gauteh, 2024-10-26:
@edwinrainville: I think the output frequency for the solution is now set to 1 Hz, does that one need to be increased if the TP frequency is increased?

gauteh, 2024-10-28:
- [x] added a on/off switch so that capacitors won't mess up voltage levels
- [ ] swapped to new batteries
- [ ] handle more panics in GPIO / RTC interrupt

gauteh, 2024-10-28:
Have had a test running from 11:43 LT without errors now @ 5 hz. will be useful for testing downloading and parsing data.

edwinrainville, 2024-10-28:
Sounds good! Just was curious!

gauteh, 2024-10-28:
I'm also working a bit on downloading collections of egps packages. I will move on to the real-time transmission in a bit (#174 ). Helge is back, he was wondering if we want him to work on the blue robotics or if you have a plan for it? He is doing a variation with PVC pipes probably.

edwinrainville, 2024-10-28:
I have a general idea of what to do with the blue robotics bottle. I will be back tonight and I think I'll just pop in and take a look at the housing and maybe we can meet up with Helge tomorrow morning and see wha the thinks?

gauteh, 2024-10-28:
Good plan! I soldered on a switch on the board (1 is off, 0 is on...), we need to solder up the next RTK buoy as well so that Helge can start working on it.

edwinrainville, 2024-10-28:
Sounds good! I'll get the second one soldered up tomorrow morning and update the firmware on it!

gauteh, 2024-10-28:
Ok!

gauteh, 2024-10-28:
[image]
This is 40 minutes of elevation from the GPS, varying. Changes about 20 cm in 40 seconds worst case (this is with a pretty poor fix, no RTK, etc).

gauteh, 2024-10-28:
Added the stationary test-data here if you want to take a look: https://github.com/gauteh/nazare/issues/5